t6x/reaver-wps-fork-t6x #321: Receive timeout occurred Sending EAPOL START request

helloful, 2020-07-11:
Hello，I got this question all the time,have any suggestion?
`[+] Waiting for beacon from 28:9E:97:AA:1F:CC
[+] Switching wlan0mon to channel 1
[+] Switching wlan0mon to channel 2
[+] Switching wlan0mon to channel 3
[+] Switching wlan0mon to channel 4
[+] Switching wlan0mon to channel 5
[+] Switching wlan0mon to channel 6
[+] Switching wlan0mon to channel 7
[+] Switching wlan0mon to channel 8
[+] Switching wlan0mon to channel 9
[+] Switching wlan0mon to channel 10
[+] Received beacon from 28:9E:97:AA:1F:CC
[+] Vendor: RalinkTe
[+] Trying pin "12345670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 28:9E:97:AA:1F:CC (ESSID: ChinaNet-J9SY)
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request

`

rofl0r, 2020-07-11:
i'd suggest to read the wiki article "troubleshooting guide" and generally moving the antenna closer to the target. let us know if you have further issues

KikoTs, 2020-08-28:
I have the same issue on TL-WN722N v1 is it the card or do i need some type of drivers because bully also doesnt work

rofl0r, 2020-09-01:
TL-WN722N v1 is one of the best cards for this purpose, but it doesn't have a lot of TX power. did you try -N command line and read the troubleshooting guide ?

soxrok2212, 2020-11-20:
Closing as no response received. Please re-open if issue persists and troubleshooting isn't helping.

closed 2020-11-20 by soxrok2212