t6x/reaver-wps-fork-t6x #322: Big_endian version pcap_open_live in -lpcap... no, error: pcap library not found!

vido89, 2020-07-12:
Hi guys, I would like to cross compile reaver package to my OpenWRT box, but Im getting this error 

```
checking for pcap_open_live in -lpcap... no
error: pcap library not found!
Makefile:54: recipe for target '/home/slobodan/lede-sdk-17.01.6-ar71xx-generic_gcc-5.4.0_musl-1.1.16.Linux-x86_64/build_dir/target-mips_24kc_musl-1.1.16/reaver-wps-fork-t6x-Big_endian/.configured_6ed3c5529d920d815e0420e8c63f7516' failed
```
But I have selected libpcap to be compiled in `make menuconfig` list and I have it installed on my machine
Any tip ? Tnx

rofl0r, 2020-07-13:
libpcap needs to be installed in the cross-compiler's library path, apparently this is not the case

vido89, 2020-07-13:
How can I add `PKG_NAME:=reaver-wps-fork-t6x` to make menuconfig list, when I do `./scripts/feeds install reaver` it's compileng old version

rofl0r, 2020-07-13:
i don't know, please ask in your distribution's help channel

adde88, 2022-01-18:
I HIGHLY recommend you cross-compile the latest commit on the master branch, instead of using the old "Big_endian" branch! 👍