t6x/reaver-wps-fork-t6x #324: Migrate from libiw to libnl3 for wireless support

nullr0ute, 2020-08-30:
The wireless extensions interface to support WiFi has been long obsolete and doesn't support a lot of the new 802.11 functionality in the kernel. There's also discussion in the kernel community about removing it as it's been long obsolete.

The new interface for dealing with 802.11 wireless interfaces on Linux is the nl80211 interface which can be used via libnl3 from a API PoV and the iw command line tool. The advantage is there's a lot more information and support for newer wireless technologies. A lot of distros are also starting to disable the old kernel wext support.

Information for iw is available here: https://wireless.wiki.kernel.org/en/users/Documentation/iw
Information for libnl3 is at: http://www.infradead.org/~tgr/libnl/

rofl0r, 2020-08-31:
thanks for the info, but your information about reaver is wrong. we use only libpcap, which wraps whatever libnl version is available on the system. additionally everything we want to achieve using reaver works already (almost) perfectly, so there's no need for any change in this area.

closed 2020-08-31 by rofl0r

nullr0ute, 2020-09-01:
You use libiw, via a bundled copy, here:
https://github.com/t6x/reaver-wps-fork-t6x/tree/master/src/lwe

And it's used here, and in a few other places, in iface.c:
https://github.com/t6x/reaver-wps-fork-t6x/blob/master/src/iface.c#L185

As it's using the kernels old Wireless Extensions interface it will cease to work when it's disabled/removed.

rofl0r, 2020-09-01:
i see. well, as far as i'm aware this is only used for things like changing the channel so it's no big deal.
we'll have to fix it once it becomes a real problem. at the moment it looks more like most ISPs disabling WPS so reaver may already have become useless once the old wireless interface has been removed from the kernel.
by that i mean removed **completely**, not only disabled by some distro maintainers. sane distros with a focus on wifi-hacking like kali won't ever disable it since it will break not only reaver but lots more of software out there.
you're of course free to propose a PR that enables the use of the new api as an optional feature, and we might merge it if it doesn't interfere with already working configurations.