t6x/reaver-wps-fork-t6x #326: Randomization of pins

4nyp0m0n05, 2020-11-20:
Could we add somethings like below? Because if pins are sequenced it could be much more slow to find pin with start 8000xxxx but if pin is 1000xxxx randomization could be much more slow. So I want to learn that randomization is a wrong way or not? And thanks for this tool people who are contributing.

defs.h
`int chs;
`
pins.c

 ```

  void shuffle(int *array, size_t n)

    {
	srand((unsigned)time(NULL)); 
    if (n > 1) 
    {
        size_t i;
        for (i = 0; i < n - 1; i++) 
        {
          size_t j = i + rand() / (RAND_MAX / (n - i) + 1);
          int t = array[j];
          array[j] = array[i];
          array[i] = t;
        }
       }
   }
      void generate_pins2()
     {	
		int array_ind[]={0,1,.....,9999};
        shuffle(array_ind,10000);
		int array_ind2[]={0,1,2,...,999};
		shuffle(array_ind2,1000);
		int i = 0, index = 0;

	
	if(get_static_p1() && !get_pin_string_mode())
	{
		i = get_k1_key_index(atoi(get_static_p1()));
		set_p1(index, k1[i].key);
		k1[i].priority = 2;
		index++;
	}
		
		for(i=0; i<P1_SIZE; i++)
		{
			if(k1[i].priority == 1)
			{
				set_p1(index, k1[i].key);
				index++;
			}
		}
        
		
		for(i=0; index < P1_SIZE; i++)
        	{
	                if(!k1[array_ind[i]].priority)
	                {
	                        set_p1(index, k1[array_ind[i]].key);
	                        index++;
			}
		}

	
	index = 0;
	if(get_static_p2() && !get_pin_string_mode())
	{
		i = get_k2_key_index(atoi(get_static_p2()));
		set_p2(index, k2[i].key);
		k2[i].priority = 2;
		index++;
	}
		
		for(i=0; i<P2_SIZE; i++)
		{
			if(k2[i].priority == 1)
			{
				set_p2(index, k2[i].key);
				index++;
			}
		}

		
        	for(i=0; index < P2_SIZE; i++)
        	{
                   if(!k2[array_ind[i]].priority)
	                {
	                        set_p2(index, k2[array_ind[i]].key);
	                        index++;
					}
			
            }

        return;
    }
```

pins.h
```
void shuffle(int*, size_t);
void generate_pins2();
```
argparse.c
``` char *short_options = "b:e:m:i:t:d:c:T:x:r:g:l:p:s:C:O:KRZA5ELfnqvDShwN6JFu";
	struct option long_options[] = {
		{ "pixie-dust", no_argument, NULL, 'K' },
		{ "randomize", no_argument, NULL, 'R' },
....
     case 'R':
                chs=1;
                break;
```
crack.c
```
        if(chs)
            generate_pins2();
	else 
	    generate_pins();
```

rofl0r, 2020-11-20:
seems like a bad idea to me, because the pin could also start with 0 or 1 and then randomization could make it much slower. apart from that, almost no routers these days are crackable with bruteforce anyways, afaik the resume functionality in .wsc files depend on the order of the pins, and last adding support for such a feature of questionable use could introduce new hard-to-detect bugs. so it's better not to.

closed 2020-11-20 by rofl0r

soxrok2212, 2020-11-20:
I think it would actually be beneficial as in last I checked, reaver tries 01234567, 00000001, 00000002.... so on and so fourth. Not very likely to have that as your pin. I think overall, it would be better to randomize (IIRC bully does this), though as @rofl0r said traditional brute force is non-existent any more so it probably wouldn't be worth the time. You can use bully though.