t6x/reaver-wps-fork-t6x #327: First half pin found, sometimes M5 message not received before NACK message

feitoi, 2021-01-06:
See output:
```
root@kali:~# reaver -i wlan1mon -b 74:EA:3A:XX:XX:XX -c 11 -vvNL

Reaver v1.6.6-git-42-gfcc9be9 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan1mon to channel 11
[?] Restore previous session for 74:EA:3A:XX:XX:XX? [n/Y] 
[+] Restored previous session
[+] Waiting for beacon from 74:EA:3A:XX:XX:XX
[+] Received beacon from 74:EA:3A:XX:XX:XX
[+] Vendor: AtherosC
[+] Trying pin "12415670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received M3 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Trying pin "12425679"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received M3 message
[+] Received M5 message
[+] Sending M6 message
[+] Received M5 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Trying pin "12420001"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin "12420001"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received M3 message
[+] Received WSC NACK
[+] Sending WSC NACK     ***** here ******
[+] Trying pin "12422227"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M3 message
[+] Sending M4 message
[+] Received M5 message
[+] Sending M6 message
[+] Received M5 message
[+] Received WSC NACK
[+] Sending WSC NACK
^C
[+] Session saved.
```

When first half pin was found and M5 message not received before NACK message then considers as **_UNKNOWN_ERROR_** and re-trying last pin?

rofl0r, 2021-01-06:
i don't understand your question.

could you please rephrase what you think is going wrong and what should happen instead?

feitoi, 2021-01-07:
First half PIN found: **1242**
```
[+] Trying pin "12415670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received M3 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Trying pin "12425679"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received M3 message
[+] Received M5 message
[+] Sending M6 message
[+] Received M5 message
[+] Received WSC NACK
[+] Sending WSC NACK
```
Received M5 message before NACK message on normal situation

But sometimes, the NACK message is received without M5 message, last message was M3
```
[+] Trying pin "12420001"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received M3 message
[+] Received M5 message
[+] Sending M6 message
[+] Received M5 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Trying pin "12422227"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received M3 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Trying pin "12423330"
```
Currently the reaver treats as **_KEY_REJECTED_** and increases to next second half pin

My question is:
Is it correct reaver to treat as **_KEY_REJECTED_** and trying next pin?
Or should reaver to treat as **_UNKNOWN_ERROR_** and re-trying last pin?

rofl0r, 2021-01-07:
> Is it correct reaver to treat as KEY_REJECTED and trying next pin?

i think it depends whether some implementors send WSC_NACK for "pin incorrect" in this situation. what happens usually when pin is incorrect at this stage? i cannot test myself right now.

feitoi, 2021-01-08:
I modified reaver for detects the situation and stop attack

```
74:EA:3A:XX:XX:XX   11  -75  1.0  Yes  AtherosC  100.0  AP-Test
80:C6:AB:XX:XX:XX    6  -68  1.0  No   Broadcom  100.0  AP-Test-2
18:D6:C7:XX:XX:XX    6  -85  1.0  No   AtherosC  100.0  AP-Test-3

root@kali:~# reaver -i wlan1mon -b 74:EA:3A:XX:XX:XX -c 11 -vNL -p 1242

Reaver v1.6.6-git-43-g2a60241 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[?] Restore previous session for 74:EA:3A:XX:XX:XX? [n/Y] n
[+] Waiting for beacon from 74:EA:3A:XX:XX:XX
[+] Received beacon from 74:EA:3A:XX:XX:XX
[+] Vendor: AtherosC
[+] Trying pin "12425679"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12420001"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12422227"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12422227"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12423330"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12424443"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] 90.95% complete @ 2021-01-07 22:06:29 (13 seconds/pin)
[+] Trying pin "12424443"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[X] CRITICAL: NACK without M5 (BELL)
[+] Quitting after 5 crack attempts
[-] Failed to recover WPA key
```

Second attack
```
root@kali:~# reaver -i wlan1mon -b 74:EA:3A:XX:XX:XX -c 11 -vNL

Reaver v1.6.6-git-43-g2a60241 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[?] Restore previous session for 74:EA:3A:XX:XX:XX? [n/Y] 
[+] Restored previous session
[+] Waiting for beacon from 74:EA:3A:XX:XX:XX
[+] Received beacon from 74:EA:3A:XX:XX:XX
[+] Vendor: AtherosC
[+] Trying pin "12425556"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12426669"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12427772"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12428885"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12429998"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12420018"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] 91.01% complete @ 2021-01-07 22:15:29 (9 seconds/pin)
[+] Trying pin "12420025"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12420032"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12420032"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12420049"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12420056"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] 91.05% complete @ 2021-01-07 22:16:32 (12 seconds/pin)
[+] Trying pin "12420063"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12420070"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12420087"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12420094"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12420100"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] 91.10% complete @ 2021-01-07 22:17:22 (11 seconds/pin)
[+] Trying pin "12420117"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Trying pin "12420124"
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[X] CRITICAL: NACK without M5 (BELL)
[+] Quitting after 16 crack attempts
[-] Failed to recover WPA key
```

Router **AP-Test** happens frequently.
I tested with **AP-Test-2**, after 200 attempts, it didn't happen, but it had happened before.
I tested with **AP-Test-3** and it didn't happen too.

feitoi, 2021-01-22:
This problem of router can also deceive the **reaver** when cracking the first half pin.
I did repeated tests for the first half pin:
```
root@kali:~# reaver -i wlan1mon -b 74:EA:3A:XX:XX:XX -c 11 -vvNL -p 1242 -g 1

Reaver v1.6.6-git-43-g2a60241 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan1mon to channel 11
[?] Restore previous session for 74:EA:3A:XX:XX:XX? [n/Y] n
[+] Waiting for beacon from 74:EA:3A:XX:XX:XX
[+] Received beacon from 74:EA:3A:XX:XX:XX
[+] Vendor: AtherosC
[+] Trying pin "12425679"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M3 message
[+] Sending M4 message
[+] Received M5 message
[+] Sending M6 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Quitting after 1 crack attempts
[-] Failed to recover WPA key

......

root@kali:~# reaver -i wlan1mon -b 74:EA:3A:XX:XX:XX -c 11 -vvNL -p 1242 -g 1

Reaver v1.6.6-git-43-g2a60241 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan1mon to channel 11
[?] Restore previous session for 74:EA:3A:XX:XX:XX? [n/Y] n
[+] Waiting for beacon from 74:EA:3A:XX:XX:XX
[+] Received beacon from 74:EA:3A:XX:XX:XX
[+] Vendor: AtherosC
[+] Trying pin "12425679"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Quitting after 1 crack attempts
[-] Failed to recover WPA key
```

And I did for the second half pin or full pin too:
```
root@kali:~# reaver -i wlan1mon -b 74:EA:3A:XX:XX:XX -c 11 -vvNL -p 12425082 -g 1

Reaver v1.6.6-git-43-g2a60241 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan1mon to channel 11
[?] Restore previous session for 74:EA:3A:XX:XX:XX? [n/Y] n
[+] Waiting for beacon from 74:EA:3A:XX:XX:XX
[+] Received beacon from 74:EA:3A:XX:XX:XX
[+] Vendor: AtherosC
[+] Trying pin "12425082"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received M5 message
[+] Sending M6 message
[+] Received M7 message
[+] Sending WSC NACK
[+] Sending WSC NACK
[+] Pin cracked in 6 seconds
[+] WPS PIN: '12425082'
[+] WPA PSK: 'xxxxxxxx123'
[+] AP SSID: 'AP-Test'

......

root@kali:~# reaver -i wlan1mon -b 74:EA:3A:XX:XX:XX -c 11 -vvNL -p 12425082 -g 1

Reaver v1.6.6-git-43-g2a60241 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan1mon to channel 11
[?] Restore previous session for 74:EA:3A:XX:XX:XX? [n/Y] n
[+] Waiting for beacon from 74:EA:3A:XX:XX:XX
[+] Received beacon from 74:EA:3A:XX:XX:XX
[+] Vendor: AtherosC
[+] Trying pin "12425082"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Quitting after 1 crack attempts
[-] Failed to recover WPA key
```
Is it because of this problem that some issues with stuck at 90.90% and stuck at 99.99%?

rofl0r, 2021-01-22:
what do you mean goes wrong here ? reaver doesn't detect that the first pin half is correct ?

feitoi, 2021-01-22:
> reaver doesn't detect that the first pin half is correct ?

yeah.

It is possible to fix the second half pin, but I don't know how to fix the first half pin.

rofl0r, 2021-01-22:
>> reaver doesn't detect that the first pin half is correct ?

> yeah.

and this happens after we already got M5 ?

feitoi, 2021-01-23:
No. In my tests, I did not restored previous session then the reaver was reseted.

rofl0r, 2021-01-23:
i don't understand a single word you're saying.

again, can you please describe precisely under which circumstances the first pin half, if it is correct, is not detected by reaver as such ? 

maybe write your reply first in your native language, then try to translate it with deepl.com. or put both your native text and the translation in the comment so i can finally make sense of what you're saying. i very much appreciate your help and technical skills, but your english is so minimalistic i have to ask 3 times for every comment you make.

feitoi, 2021-01-24:
Sorry for my BAD english.

I found a router, it stay WPS lock after 10 attempts, but it can be used with parameter -L, and it stay "_true WPS lock_" after 200 attempts, this case needs a reboot.
Many days later, I found the first half pin, while trying to find the second half pin, I saw that sometimes the router sends NACK without sending M5 before.
Look the output:
```
root@kali:~# reaver -i wlan1mon -b 74:EA:3A:XX:XX:XX -c 11 -vvNL

Reaver v1.6.6-git-42-gfcc9be9 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan1mon to channel 11
[?] Restore previous session for 74:EA:3A:XX:XX:XX? [n/Y] 
[+] Restored previous session
[+] Waiting for beacon from 74:EA:3A:XX:XX:XX
[+] Received beacon from 74:EA:3A:XX:XX:XX
[+] Vendor: AtherosC
[+] Trying pin "12415670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received M3 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Trying pin "12425679"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received M3 message
[+] Received M5 message
[+] Sending M6 message
[+] Received M5 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Trying pin "12420001"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received M3 message
[+] Received WSC NACK
[... 147 lines cut ...]
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Trying pin "12345083"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 74:EA:3A:XX:XX:XX (ESSID: AP-Test)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Quitting after 2 crack attempts
[-] Failed to recover WPA key
```

The problem/bug is on the side of the router. There is no pattern for the problem to appear, it is random. I believe there was a processing error in the router and the router simply sends a NACK to signal the end of authentication.

feitoi, 2021-05-27:
Fixed by 4b921d1, 387678d and d6c931c.

closed 2021-05-27 by feitoi