t6x/reaver-wps-fork-t6x #331: Feature Request: Macchanger implementation

elleven11, 2021-02-11:
Every once in a while I find myself swapping reaver to the patched reaver fork that has the --macchanger option.
It would be better if this fork had it.

Here is the source of the reaver im talking about:
https://github.com/gabrielrcouto/reaver-wps

rofl0r, 2021-02-11:
you could help by diffing his version recursively against reaver-1.4, then posting the resulting patch here so we can have a look how it's implemented.

elleven11, 2021-02-12:
Sure! Where can i find the code for 1.4? 
the link in the readme for the source archive is dead (https://code.google.com/p/reaver-wps/)

rofl0r, 2021-02-12:
either git checkout de276e88c4653eeef0bf40e03c5d3c8bb65cabbb on this repo, or  http://reaver-wps.googlecode.com/files/reaver-1.4.tar.gz

rofl0r, 2021-04-15:
merged via https://github.com/t6x/reaver-wps-fork-t6x/pull/336

closed 2021-04-15 by rofl0r