t6x/reaver-wps-fork-t6x #333: RTL8812BU  Timeout

wukuy, 2021-02-16:
```bash
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
```

```bash
sudo reaver -i wlan0 -b C8:E7:D8:17:83:AC -S -d2 -t2 -vvv -N

Reaver v1.6.6-git-2-g2d6e537 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Waiting for beacon from C8:E7:D8:17:83:AC
[+] Switching wlan0 to channel 1
[+] Switching wlan0 to channel 2
[+] Switching wlan0 to channel 1
[+] Received beacon from C8:E7:D8:17:83:AC
[+] Vendor: Unknown 
WPS: A new PIN configured (timeout=0)
WPS: UUID - hexdump(len=16): [NULL]
WPS: PIN - hexdump_ascii(len=8):
     31 32 33 34 35 36 37 30                           12345670        
WPS: Selected registrar information changed
WPS: Internal Registrar selected (pbc=0)
WPS: sel_reg_union
WPS: set_ie
WPS: cb_set_sel_reg
WPS: Enter wps_cg_set_sel_reg
WPS: Leave wps_cg_set_sel_reg early
WPS: return from wps_selected_registrar_changed
[+] Trying pin "12345670"
send_packet called from deauthenticate() 80211.c:380
send_packet called from authenticate() 80211.c:411
[+] Sending authentication request
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[!] WARNING: Receive timeout occurred
send_packet called from deauthenticate() 80211.c:380
send_packet called from authenticate() 80211.c:411
[+] Sending authentication request
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[!] WARNING: Receive timeout occurred
send_packet called from deauthenticate() 80211.c:380
send_packet called from authenticate() 80211.c:411
[+] Sending authentication request
send_packet called from associate() 80211.c:464
[+] Sending association request
send_packet called from resend_last_packet() send.c:161
[+] Associated with C8:E7:D8:17:83:AC (ESSID: 502)
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
^Csend_packet called from send_termination() send.c:142

```

```bash
pi@raspberrypi:~ $ sudo aireplay-ng -9 -a C8:E7:D8:17:83:CC -e "592" wlan0 -D
02:10:16  Trying broadcast probe requests...
02:10:17  No Answer...
02:10:17  Found 1 AP 

02:10:17  Trying directed probe requests...
02:10:17  C8:E7:D8:17:83:AC - channel: 0 - '502'
02:10:23  Ping (min/avg/max): 6.480ms/14.400ms/22.320ms Power: -67.00
02:10:23   2/30:   6%

02:10:23  Injection is working!
```

[output.cap.gz](https://github.com/t6x/reaver-wps-fork-t6x/files/5990819/output.cap.gz)

rofl0r, 2021-02-16:
what is your question ?

kcdtv, 2022-03-11:
I do not know but i guess it is something like "Why reaver does not work?"
And yes, "injection is working" but... ...Check the rate: `**02:10:23   2/30:   6%**`
It is impossible to do anything with such a low rate, 
Are you using the internal wifi chip of the raspberry pi?
Which version of pi is it?
Can we get the stdout for `sudo airmoin.ng `?