t6x/reaver-wps-fork-t6x #334: Reaver can't crack wi-fi stuck on the same pin (12345670)

BasselSabour, 2021-02-25:
I'm trying to crack my own wi-fi through WPS using reaver using `sudo reaver -i wlan0 -b 84:9F:B5:71:48:28 -vv` and it keeps outputting these messages repeatedly. I'm using tp link wn722n wireless adapter v3. It supports monitor mode and packet injection (after a bit of exhaustion). There's one thing that doesn't work on this adapter which is changing its mac address (don't know if that would be useful or not).

I left it for 10 hours and it's still like that.

The problem is the with the error (WPS transaction failed (code: 0x02), re-trying last pin) and that it's stuck on the same pin (12345670)

P.S: I'm a bit of a newbie to linux, so please make your answer detailed.
`[+] Waiting for beacon from 20:E8:82:AE:36:76
[+] Switching wlan0 to channel 1
[+] Received beacon from 20:E8:82:AE:36:76
[+] Vendor: RealtekS
[+] Trying pin "12345670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 20:E8:82:AE:36:76 (ESSID: WE_AE3676)
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[!] WARNING: 10 successive start failures
[+] Sending EAPOL START request
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin`

rofl0r, 2021-02-25:
well, you don't even get past the association step. check the troubleshooting guide here on the wiki

BasselSabour, 2021-02-27:
where is it? @rofl0r

rofl0r, 2021-02-27:
click on wiki https://github.com/t6x/reaver-wps-fork-t6x/wiki then on the menu on the right on troubleshooting https://github.com/t6x/reaver-wps-fork-t6x/wiki/Troubleshooting

closed 2021-02-27 by rofl0r