t6x/reaver-wps-fork-t6x #337: send_packet called from send_eapol_start()

dimk1, 2021-04-15:
Hello there,

running the latest 32-bit Kali version on the latest RPi4. Reaver consistently fails with the following output:

```
# reaver --bssid TA:RG:ET:AD:RR:ES --channel 11 --interface wlan0mon -vvv --no-associate

Reaver v1.6.6 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan0mon to channel 11
[+] Waiting for beacon from TA:RG:ET:AD:RR:ES
[+] Received beacon from TA:RG:ET:AD:RR:ES
[+] Vendor: AtherosC
WPS: A new PIN configured (timeout=0)
WPS: UUID - hexdump(len=16): [NULL]
WPS: PIN - hexdump_ascii(len=8):
     31 32 33 34 35 36 37 30                           12345670        
WPS: Selected registrar information changed
WPS: Internal Registrar selected (pbc=0)
WPS: sel_reg_union
WPS: set_ie
WPS: cb_set_sel_reg
WPS: Enter wps_cg_set_sel_reg
WPS: Leave wps_cg_set_sel_reg early
WPS: return from wps_selected_registrar_changed
[+] Trying pin "12345670"
[+] Associated with TA:RG:ET:AD:RR:ES (ESSID: target_AP)
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
[+] Received deauth request
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
```
aireplay creates a correct association
```
# aireplay-ng --fakeauth 30 -a TA:RG:ET:AD:RR:ES -h MY:MA:CA:DD:RE:SS wlan0mon
19:35:12  Waiting for beacon frame (BSSID: TA:RG:ET:AD:RR:ES) on channel 11

19:35:12  Sending Authentication Request (Open System) [ACK]
19:35:12  Authentication successful
19:35:12  Sending Association Request [ACK]
19:35:13  Association successful :-) (AID: 1)
```
Injection also works
```
# aireplay-ng -9 -a TA:RG:ET:AD:RR:ES -e "target_AP" wlan0mon
19:40:46  Waiting for beacon frame (BSSID: TA:RG:ET:AD:RR:ES) on channel 11
19:40:46  Trying broadcast probe requests...
19:40:46  Injection is working!
19:40:48  Found 1 AP 

19:40:48  Trying directed probe requests...
19:40:48  TA:RG:ET:AD:RR:ES - channel: 11 - 'target_AP'
19:40:52  Ping (min/avg/max): 2.334ms/33.941ms/42.813ms Power: -63.12
19:40:52   8/30:  26%
```
I use the built-in wifi in the RPi
```
# airmon-ng start wlan0

PHY	Interface	Driver		Chipset

phy0	wlan0		brcmfmac	Broadcom 43430
		(mac80211 monitor mode vif enabled for [phy0]wlan0 on [phy0]wlan0mon)
		(mac80211 station mode vif disabled for [phy0]wlan0)
```
I also captured a wireshark tracer. After the EAPOL START message the target replies with a Deauthentication which also appears in the reaver output. A bug or simply the attack will fail because of the configuration of the target?

rofl0r, 2021-04-15:
> [+] Received deauth request

yes, that's the cause. we haven't decided yet what the best way to proceed is when that happens so for the moment we opted to just print it so the user can decide on his own. though that's certainly not optimal especially in bruteforce mode. but then, using external association we have no way to tell what the current state of the connection is.

maybe we should just remove the no-associate option or replace it with a call to aircrack-ng... but then i dont think it's necessary as the associate code seems to be working quite well lately.

dimk1, 2021-04-16:
Οk, I will give it a try with reaver's association option and inform back. Since I'm just starting to study ethical hacking, what are possible reasons for this deauthentication reply?

rofl0r, 2021-04-16:
well for example aircrack-ng sending assoc request while reaver already sends eapol, so router gets confused and wants to start over. it could also be some sort of wps hack protection.

soxrok2212, 2022-10-21:
Closing as no discussion for over a year.

closed 2022-10-21 by soxrok2212