t6x/reaver-wps-fork-t6x #339: it is Showing Timeout error

shreesha345, 2021-04-25:
when I start this command 
`sudo reaver --bssid C4:E9:84:8F:ED:84 --channel 3 --interface wlan0 -vv` and I am using tp-link version v2/v3 so it is enabled monitor mode and th name is wlan0

so Now the issue 
```
Reaver v1.6.6 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan0 to channel 3
[+] Waiting for beacon from C4:E9:84:8F:ED:84
[+] Received beacon from C4:E9:84:8F:ED:84
[+] Vendor: AtherosC
[+] Trying pin "12345670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with C4:E9:84:8F:ED:84 (ESSID: ROYAL GIRLS HOSTEL2)
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
^C
[+] Nothing done, nothing to save.
```
I am targeting WPS so please fix this issue

rofl0r, 2021-04-25:
fix *your* issues and read the troubleshooting guide here in the wiki.

closed 2021-04-25 by rofl0r

shreesha345, 2021-04-26:
can you help me I reared the troubleshooting  but it did not work please help me @rofl0r