t6x/reaver-wps-fork-t6x #351: wash --pbc ?

minanagehsalalma, 2022-01-22:
I am looking for a way to if a network has its wps button pushed 
without trying to connect to it but with just monitoring it 

is that possible ? 

It would be a really cool addition to WASH if it can tell the networks whom buttons was pushed 
and print it in scan with timeout or something 

Any way to do this would be great 
As i am trying mix it with Phishing , instead of the password.

Thanks

soxrok2212, 2022-01-22:
It is possible and pixiewps supports it for old Realtek chips that used a static diffie-hellman key. I think arg -7 is what you need. You can get a passive capture with airodump and grab the parameters from the capture.

soxrok2212, 2022-01-22:
Outside of seeing it in your captures, there’s no tool that will tell you if it sees PBC activated.

minanagehsalalma, 2022-01-22:
> It is possible and pixiewps supports it for old Realtek chips that used a static diffie-hellman key. I think arg -7 is what you need. You can get a passive capture with airodump and grab the parameters from the capture.

Oh ,Thanks a lot for your reply mate 

But what are parameters to look for ? 

and would that method be fast enough To be able to get to start a connection when the parameters are detected as i mean WPS PBC has `120 ` sec timeout

soxrok2212, 2022-01-22:
It doesn't really matter because you can decrypt all messages. So you can get the PIN and the PSK. Check the README for a quick example.

closed 2022-01-22 by soxrok2212

minanagehsalalma, 2022-01-22:
> It doesn't really matter because you can decrypt all messages. So you can get the PIN and the PSK. Check the README for a quick example.

@soxrok2212 I checked it but i can't still get how to use the ` arg -7 ` ?
what is the syntax for it ? 

I have to collect the 
`--pke ... --pkr ... --e-hash1 ... --e-hash2 ... --authkey ... --e-nonce ...`

first ? 

then use them in addition to arg -7 ?

Thanks 

i still want to know search filter for airodump to know if it's pressed or not 

also are you available for chatting ? as i want to add this to a more advance attack ... A phishing type

minanagehsalalma, 2022-01-22:
I tried it use -7 arg 
but it says 
`pixiewps: option requires an argument -- '7'
`

So , How do i actually use it ?

rofl0r, 2022-01-22:
second picture in pixiewps readme:
https://camo.githubusercontent.com/18d30ce7809015bae46a7b5e10957e762e269e01a26ab7b8f7accb146520c1f8/68747470733a2f2f692e696d6775722e636f6d2f71565138526e672e706e67

rofl0r, 2022-01-25:
ftr this discussion moved to https://github.com/wiire-a/pixiewps/issues/107