t6x/reaver-wps-fork-t6x #353: quick fix maybe

ChillVibesMushroom, 2022-02-28:
installed correctly so I'm just trying to crack into my router that is wps enabled and an AtherosC router

command used

sudo reaver -i wlan1mon -b 00:7F:28:19:0B:31   -d 7 -vv -N 

error 

Found packet with bad FCS, skipping...

error 

WARNING: Receive timeout occurred

error

WPS transaction failed (code: 0x03), re-trying last pin

error

Received EAP_FAILURE message

output

[+] Waiting for beacon from 00:7F:28:19:0B:31
[!] Found packet with bad FCS, skipping...
[+] Switching wlan1mon to channel 1
[+] Switching wlan1mon to channel 2
[+] Switching wlan1mon to channel 3
[+] Switching wlan1mon to channel 4
[+] Switching wlan1mon to channel 5
[+] Switching wlan1mon to channel 6
[+] Received beacon from 00:7F:28:19:0B:31
[+] Vendor: AtherosC
[+] Trying pin "12345670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 00:7F:28:19:0B:31 (ESSID: 9LBC4)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin "12345670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 00:7F:28:19:0B:31 (ESSID: 9LBC4)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received EAP_FAILURE message
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x03), re-trying last pin
[+] Trying pin "12345670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 00:7F:28:19:0B:31 (ESSID: 9LBC4)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received EAP_FAILURE message
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x03), re-trying last pin
[+] Trying pin "12345670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 00:7F:28:19:0B:31 (ESSID: 9LBC4)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received EAP_FAILURE message
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x03), re-trying last pin
[+] Trying pin "12345670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 00:7F:28:19:0B:31 (ESSID: 9LBC4)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received EAP_FAILURE message
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x03), re-trying last pin
[+] Trying pin "12345670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 00:7F:28:19:0B:31 (ESSID: 9LBC4)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received EAP_FAILURE message
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x03), re-trying last pin
[+] Nothing done, nothing to save.
[+] 0.00% complete @ 2022-02-28 13:43:06 (0 seconds/pin)
[+] Trying pin "12345670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 00:7F:28:19:0B:31 (ESSID: 9LBC4)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received EAP_FAILURE message
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x03), re-trying last pin
[+] Trying pin "12345670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 00:7F:28:19:0B:31 (ESSID: 9LBC4)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received EAP_FAILURE message
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x03), re-trying last pin
[+] Trying pin "12345670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 00:7F:28:19:0B:31 (ESSID: 9LBC4)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received EAP_FAILURE message
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x03), re-trying last pin
^C
[+] Nothing done, nothing to save.

kcdtv, 2022-03-11:
**-91** is too weak. 
Get closer to the AP,

soxrok2212, 2022-10-21:
No response in months, closing for now.

closed 2022-10-21 by soxrok2212