t6x/reaver-wps-fork-t6x #364: Does wash will detect 6GHz in the future  ?

Koala633, 2022-10-21:
Hi,

It's not an issue but just a question, does wash will be maintained to detect networks on 6GHz frequency ?

Today the wifi 6E is alive in most of modern AP and the wifi 7 will coming out of 2024 (if all things go well with the wifi alliance).So i just need to know how wash will work with wifi 6E or 7.

Thanks a lot for made reaver and wash :)

soxrok2212, 2022-10-21:
> It's not an issue but just a question, does wash will be maintained to detect networks on 6GHz frequency ?

IMO, probably not. I suppose it might not be hard to allow scanning of 6GHz channels, but it's going to depend on:

- Monitor mode support on a 6E chip
- ... of which I'm not even aware of any yet
- WPS still being enabled on those bands
- ... of which I don't think its required to be Wi-Fi certified now, not to mention WPS is basically dead on access points 802.11ac or newer (exception is range extenders which rarely see updates and are often based on old reference code).

Koala633, 2022-10-21:
Ok i got it ! thanks for you answer and all your work with reaver and wash, you let a mark in the history of the wifi penetration testing :)

closed 2022-10-21 by Koala633