t6x/reaver-wps-fork-t6x #365: BUG (?) Segmentation Fault

vincenzogianfelice, 2023-01-19:
**Debian 11** (5.10), **AWUS036AC** (rtl8812au), **iwlwifi**
```
[+] Trying pin "12345670"
send_packet called from deauthenticate() 80211.c:380
send_packet called from authenticate() 80211.c:411
[+] Sending authentication request
send_packet called from associate() 80211.c:464
[+] Sending association request
[+] Associated with D6:35:1D:5D:B4:B4 (ESSID: <ESSID>)
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
Segmentation fault
```

This problem after latest **git pull**.

rofl0r, 2023-02-01:
try `make clean` before `make`. if the problem persists, use CFLAGS to activate your compiler's address sanitizer, so when it crashes again it will tell you (and us via your paste) exactly where it happens.

alexandrglm, 2023-10-21:
Did no one solved this?

rofl0r, 2023-10-30:
we don't even have confirmation from OP whether it's a real bug, not lack of running make clean, much less a backtrace or address sanitizer analysis output. until this happens, consider it PEBKAC.