t6x/reaver-wps-fork-t6x #368: reaver getting block ack

Silentkiller987, 2023-03-24:
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[!] WARNING: Receive timeout occurred
[!] WARNING: 10 successive start failures
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
WPS: Building Message WSC_NACK
WPS:  * Version
WPS:  * Message Type (14)
WPS:  * Enrollee Nonce
WPS:  * Registrar Nonce
WPS:  * Configuration Error (0)
[+] Sending WSC NACK
send_packet called from send_msg() send.c:116
[!] WPS transaction failed (code: 0x02), re-trying last pin
WPS: A new PIN configured (timeout=0)
WPS: UUID - hexdump(len=16): [NULL]
WPS: PIN - hexdump_ascii(len=8):
     31 32 33 34 35 36 37 30                           12345670        
WPS: Selected registrar information changed
WPS: Internal Registrar selected (pbc=0)
WPS: sel_reg_union
WPS: set_ie
WPS: cb_set_sel_reg
WPS: Enter wps_cg_set_sel_reg
WPS: Leave wps_cg_set_sel_reg early
WPS: return from wps_selected_registrar_changed
[+] Trying pin "12345670"
[+] Associated with FC:22:F4:2C:A4:54 (ESSID: Airtel_9660552532)
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[+] Received deauth request
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[... 84 lines cut ...]
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161

soxrok2212, 2023-03-24:
What is the model of the AP? What is the rssi?

Silentkiller987, 2023-03-24:
> What is the model of the AP? What is the rssi?

zyxel

Silentkiller987, 2023-03-24:
> What is the model of the AP? What is the rssi?

WPS transaction failed (code: 0x02), re-trying last pin
[+] Nothing done, nothing to save.
[+] 0.00% complete @ 2023-03-24 11:56:49 (0 seconds/pin)
[+] Trying pin "12345670"
[+] Associated with FC:22:F4:2C:A4:54 (ESSID: Airtel_9660552532)
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Received M1 message
[+] Received M1 message
[+] Received M1 message
[+] Received M1 message
[+] Received M1 message
[+] Received M1 message
[+] Received M1 message
[+] Received M1 message
[+] Received M1 message
[+] Received M1 message
[+] Received M1 message
[+] Received M1 message
[+] Received M1 message
[+] Received M1 message
[+] Received M3 message
[+] Sending M4 message
[+] Received M3 message
[+] Received M3 message
[+] Received M3 message
[+] Received M3 message
[+] Received M3 message
[+] Received M3 message
[+] Received M3 message
[+] Received M3 message
[+] Received M3 message
[+] Received M3 message
[+] Received M3 message
[+] Received M3 message
[+] Received M3 message
[+] Received M3 message
[+] Received M3 message
[+] Received M3 message
[+] Received M5 message
[+] Sending M6 message
[+] Received M5 message
[+] Received M5 message
[+] Received M5 message
[+] Received M5 message
[+] Received M5 message
[+] Received M5 message
[+] Received M5 message
[+] Received M5 message
[+] Received M5 message
[+] Received M5 message
[+] Received M5 message
[+] Received M5 message
[+] Received M5 message
[+] Received M5 message
[+] Received M5 message
[+] Received M5 message
[+] Received M7 message
[+] Sending WSC NACK
[+] Sending WSC NACK
[+] Pin cracked in 835 seconds
[+] WPS PIN: '12345670'
[+] WPA PSK: 'Jain80035.35435Malti'
[+] AP SSID: 'Airtel_9660552532'
 

when it is getting successful i am getting the old password

soxrok2212, 2023-03-24:
Not the manufacturer, the actual model. And if you’re getting the old PSK, that’s a problem on the router, not reaver.

Silentkiller987, 2023-03-24:
> Not the manufacturer, the actual model. And if you’re getting the old PSK, that’s a problem on the router, not reaver.

so what should i do to fix it any suggestions

soxrok2212, 2023-03-24:
Provide us with the information requested

Silentkiller987, 2023-03-24:
> Provide us with the information requested

pcap file ??

Silentkiller987, 2023-03-24:
> > Provide us with the information requested
> 
> pcap file ??

can i contact you on discord and share you the details

soxrok2212, 2023-03-24:
No, the router model

Silentkiller987, 2023-03-24:
> No, the router model

router model number is zyxel PMG5617-R20B

soxrok2212, 2023-03-24:
Have you tried changing the router password, running reaver again and seeing if it grabs the right key?

Silentkiller987, 2023-03-24:
> Have you tried changing the router password, running reaver again and seeing if it grabs the right key?

i did first when i got the password i changed the password and retry it but it was showing me the old password then i checked the pcap it was showing block ack then i associated using aireplay-ng but after that it started showing me the old password so i thought something might be with reaver so i used bully but bully lck the router then i used mdk4 to reset it but after that now i can't even capture handshake also

feitoi, 2023-03-27:
> i did first when i got the password i changed the password and retry it but it was showing me the old password

Does your router have 2.4Ghz and 5Ghz? Did you change the password for both?

Silentkiller987, 2023-03-27:
> > i did first when i got the password i changed the password and retry it but it was showing me the old password
> 
> Does your router have 2.4Ghz and 5Ghz? Did you change the password for both?

just 2.4ghz but still i didn't get why i can't capture handshake also

Silentkiller987, 2023-03-27:
just 2.4ghz but still i didn't get why i can't capture handshake also

october262, 2023-03-28:
try with reaver -i wlan0 -b XX:XX:XX:XX:XX:XX -vv -p 12345670
and see if you get the correct password / PSK
maybe your not saving the changes when you close the configuring page settings.

Silentkiller987, 2023-03-28:
Noo i checked it and my devices are not connecting also with the old
password

october262, 2023-03-29:
once you have the password / PSK changed,
try unplugging the router from the wall outlet and let it sit for about 20 minutes
and then see if the new password / PSK sticks.

Silentkiller987, 2023-03-29:
Ok so in simple words i just need to restart the router

closed 2024-07-26 by soxrok2212