t6x/reaver-wps-fork-t6x #371: Reaver session gets stuck at "send_packet called from resend_last_packet() send.c:161"

mkarne1, 2023-04-21:
Hi there, 

my reaver session always gets stuck at "send_packet called from resend_last_packet() send.c:161" while cracking my WPS network with the following settings:

[image]

**I'm using:**

Reaver version: 1.6.6 
Wireless Adapter: Alfa AWUS036ACH
Host OS: macOS BigSur
VMware Fusion with: Linux kali 6.1.0-kali7-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.20-1kali1 (2023-03-22) x86_64 GNU/Linux

**Here are the commands that I'm running + their output:**

Thanks in advance & best regards!

````
──(root㉿kali)-[~]
└─# aireplay-ng --fakeauth 100 -a D4:3D:F3:XX:XX:XX -h EA:E1:A1:XX:XX:XX wlan0
11:39:20  Waiting for beacon frame (BSSID: D4:3D:F3:XX:XX:XX1) on channel 8

11:39:20  Sending Authentication Request (Open System) [ACK]
11:39:20  Authentication successful
11:39:20  Sending Association Request [ACK]
11:39:20  Association successful :-) (AID: 1)

11:39:34  Got a deauthentication packet! (Waiting 3 seconds)

11:39:37  Sending Authentication Request (Open System) [ACK]
11:39:38  Got a deauthentication packet! (Waiting 5 seconds)

11:39:43  Sending Authentication Request (Open System) [ACK]
11:39:43  Authentication successful
11:39:43  Sending Association Request [ACK]
11:39:43  Got a deauthentication packet! (Waiting 7 seconds)

11:39:50  Sending Authentication Request (Open System) [ACK]
11:39:50  Got a deauthentication packet! (Waiting 9 seconds)

11:39:59  Sending Authentication Request (Open System) [ACK]
11:39:59  Got a deauthentication packet! (Waiting 11 seconds)

11:40:10  Sending Authentication Request (Open System) [ACK]
11:40:10  Got a deauthentication packet! (Waiting 13 seconds)

11:40:23  Sending Authentication Request (Open System) [ACK]
11:40:24  Authentication successful
11:40:24  Sending Association Request [ACK]
11:40:24  Association successful :-) (AID: 1)

11:40:39  Sending keep-alive packet [ACK]
11:40:54  Sending keep-alive packet [ACK]
11:41:09  Sending keep-alive packet [ACK]^C

──(root㉿kali)-[~]
└─# reaver --bssid D4:3D:F3:XX:XX:XX --channel 8 --interface wlan0 -vvv --no-associate

Reaver v1.6.6 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan0 to channel 8
[?] Restore previous session for D4:3D:F3:XX:XX:XX? [n/Y] n
[+] Waiting for beacon from D4:3D:F3:XX:XX:XX
[+] Received beacon from D4:3D:F3:XX:XX:XX
[+] Vendor: RalinkTe
WPS: A new PIN configured (timeout=0)
WPS: UUID - hexdump(len=16): [NULL]
WPS: PIN - hexdump_ascii(len=8):
     31 32 33 34 35 36 37 30                           12345670        
[... 338 lines cut ...]
[+] Received deauth request
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
^Csend_packet called from send_termination() send.c:142

[+] Session saved.

soxrok2212, 2023-04-21:
Try with the -N flag

rofl0r, 2023-04-21:
we haven't yet decided how to handle deauth requests. i guess in bruteforce mode it would be best to send deauth, and start over (including auth, assoc) with the current pin.

mkarne1, 2023-04-21:
> we haven't yet decided how to handle deauth requests. i guess in bruteforce mode it would be best to send deauth, and start over (including auth, assoc) with the current pin.

@rofl0r thanks for the quick reply! 

I'm not sure if I fully understand your reply. Is there currently a way to solve this issue? Or is this a known issue that has no fix yet?

rofl0r, 2023-04-21:
> I'm not sure if I fully understand your reply. Is there currently a way to solve this issue? Or is this a known issue that has no fix yet?

it's a TODO item, so it needs someone to implement handling deauth request in bruteforce mode. the only contributor i'm aware of that still uses bruteforce mode in the age of WPS 2.0 is @feitoi . maybe he can be bothered to implement it.

mkarne1, 2023-04-21:
@rofl0r Okay got it, then I think we can close this issue. Just one last question: Is there any other / better / recommend way of cracking WPS networks?

rofl0r, 2023-04-21:
the bruteforce method is almost useless these days, the prefered method is to identify the algorithm used by the vendor to generate the wps pin (often from mac address), calculate the right pin in advance and use reaver or oneshot to send that single pin. [3wifi](http://3wifi.stascorp.com/wpspin) has a website with an online keygen for many router models. that or pixiewps, if the router is vulnerable.

mkarne1, 2023-04-22:
@rofl0r Thanks a lot, I really appreciate your help!

closed 2023-04-22 by mkarne1

soxrok2212, 2023-04-22:
Thank you @mkarne1 for being patient and polite. It’s hard to come by these days. In the mean time, you could try the tool “bully” but I’m unsure of the state of that one too.

mkarne1, 2023-04-22:
@soxrok2212 nothing to thank me for, you guys are providing all of us with a great, free tool and provide great, free support on top. The least I can be is friendly! :)

mkarne1, 2023-04-24:
> the bruteforce method is almost useless these days, the prefered method is to identify the algorithm used by the vendor to generate the wps pin (often from mac address), calculate the right pin in [...]

@rofl0r sorry to bother you again, but do you happen to have an invite code? The WPS PIN generator is only accessible for members and registration is limited to those who have a code. 

According to [this post](https://forum.antichat.com/threads/427654/page-104#post-4144869) I can buy it from [binarymaster](https://forum.antichat.com/members/148032/), but he wasn't online for a few months and I'm not sure if/when he will read my message. 

Thanks for your time & best regards.

rofl0r, 2023-04-24:
@feitoi 

> [+] Received deauth request

problem is with routers that send deauth request during a WPS tx. reaver currently doesn't handle it at all which makes bruteforce mode stuck.

@mkarne1 the wpspin site has 2 forms, the right one (database search) is only for members, the left one available for everyone. it uses javascript to generate pins with the selected method from the mac address you enter. i don't have an invitation code for the site.