t6x/reaver-wps-fork-t6x #376: Reaver stuck on send_packet called from resend _last_packet() send.c.161

Abhinav413652, 2023-08-13:
[image]

I've never successfully executed reaver. I'm trying the pixie dust attack but it always fails. I've cloned the repo successfully but reaver always gives me the same output. I've tried multiple commands but nothing seems to work

soxrok2212, 2023-08-13:
What is the model of the AP? Is WPS enabled and not locked?

Abhinav413652, 2023-08-15:
AP is of RaLink and WPS is enabled and is not locked.

Abhinav413652, 2023-08-15:
Btw I closed this by mistake.

october262, 2023-08-15:
if you try your reaver command again without the pixie dust command
are you getting  Received M1 message Sending M2 message Received M3 message
Sending M4 message ??

rofl0r, 2023-08-16:
OP doesnt even get associated according to screenshot, so he's probably too far away or his dongle doesnt support injection. read the Troubleshooting article on the wiki here.

Abhinav413652, 2023-08-17:
I'm not far, I'm very close to the target

soxrok2212, 2023-08-17:
What happens if you stand right next to it?

Abhinav413652, 2023-08-17:
My wifi adapter supports packet injection as it can switch to monitor mode. I was watching a lecture and he mentioned that latest versions of reaver have this error. Is it so?

october262, 2023-08-17:
try reaver like this - reaver -i wlan0 -b F2:ED:B8:5D:84:17 -vv -P your routers pin #
see if it work this way.

Abhinav413652, 2023-08-17:
I'll try tomorrow thx 🙏 🙏 for the response

Abhinav413652, 2023-08-19:
I tried and still it's giving me the same error

rofl0r, 2023-08-19:
> My wifi adapter supports packet injection as it can switch to monitor mode

that's a wrong conclusion. some drivers support one, but not the the other, even if not by design then by kernel bugs. make the injection test according to the troubleshooting guide and post the result. my patience is about to run out.

Abhinav413652, 2023-08-20:
see my adapter can do packet injection as it did crack TP LINK wps using my adapter using wifite.
is there any tutorial which works now?? as wifite and airegeddon are also failing...

soxrok2212, 2023-08-20:
Unable to provide necessary information, cannot prove ownership of target AP.

closed 2023-08-20 by soxrok2212