t6x/reaver-wps-fork-t6x #379: Rstoring the PIN for my router, always fails!

reaveruser, 2024-05-04:
It is my router, I have the WIFI code and I can access the web based setting. If I either set the router to WPS PBC or PIN code, it will only show the PIN option when I try to connect to the router.
It is huawei b593u-12 and airodump-nh show the WPS version is 1.0 and it is not locked. I even tried to rest the router and it didn't help

Reaver keeps showing error 0x04 sometimes 0x3 and it fails to test any PIN. The router is only 30 CM away from the wifi adapter.
The reaver attack works with other models, except my router.

Any way I can authenticate with the actual correct WIFI password as I know it and I can't find any option to take advantage of knowing the password?

If you need any information to help me please ask, I am willing to provide anything.

rofl0r, 2024-05-05:
provide wash -j output for the router, and the log output from reaver (you can redact the mac address and network name)

reaveruser, 2024-05-06:
Any way to take advantage of the WIFI password as I know it?

Here is the wash and log files for reaver:
Wash:

> ─$ sudo wash -i wlan0mon -c 6 -j {"bssid" : "xx:xx:xx:xx:xx:xx", "essid" : "", "channel" : 6, "rssi" : -69, "vendor_oui" : "001018", "wps_version" : 16, "wps_state" : 2, "wps_locked" : 2, "wps_model_number" : "123456", "wps_serial" : "1234", "wps_uuid" : "00000000000000000000000000000000", "wps_response_type" : "03", "wps_primary_device_type" : "00060050f2040001", "wps_config_methods" : "0000", "wps_rf_bands" : "01", "dummy": 0} 

Reaver:

> └─$ sudo reaver -i wlan0mon -c 6 -b xx:xx:xx:xx:xx:xx -vv
> 
> Reaver v1.6.6 WiFi Protected Setup Attack Tool
> Copyright (c) 2011, Tactical Network Solutions, Craig Heffner cheffner@tacnetsol.com
> 
> [+] Switching wlan0mon to channel 6
> [?] Restore previous session for xx:xx:xx:xx:xx:xx? [n/Y] n
> [+] Waiting for beacon from xx:xx:xx:xx:xx:xx
> [+] Received beacon from xx:xx:xx:xx:xx:xx
> [+] Vendor: Broadcom
> [+] Trying pin "12345670"
> [+] Sending authentication request
> [+] Sending association request
> [+] Associated with xx:xx:xx:xx:xx:xx (ESSID: )
> [+] Sending EAPOL START request
> [+] Received identity request
> [+] Sending identity response
> [+] Received deauth request
> [!] WARNING: Receive timeout occurred
> [+] Sending WSC NACK
> [!] WPS transaction failed (code: 0x02), re-trying last pin
> [+] Trying pin "12345670"
> [+] Sending authentication request
> [+] Sending association request
> [+] Associated with xx:xx:xx:xx:xx:xx (ESSID: )
> [+] Sending EAPOL START request
> [+] Received identity request
> [+] Sending identity response
> [+] Received deauth request
> [!] WARNING: Receive timeout occurred
> [+] Sending WSC NACK
> [!] WPS transaction failed (code: 0x02), re-trying last pin
> [+] Trying pin "12345670"
> [+] Sending authentication request
> [+] Sending association request
> [+] Associated with xx:xx:xx:xx:xx:xx (ESSID: )
> [+] Sending EAPOL START request
> [+] Received identity request
> [+] Sending identity response
> [+] Received M1 message
> [+] Sending M2 message
> [+] Received WSC NACK
> [+] Sending WSC NACK
> [!] WPS transaction failed (code: 0x04), re-trying last pin
> [+] Trying pin "12345670"
> [+] Sending authentication request
> [+] Sending association request
> [+] Associated with xx:xx:xx:xx:xx:xx (ESSID: )
> [+] Sending EAPOL START request
> [+] Received identity request
> [+] Sending identity response
> [+] Received deauth request
> [!] WARNING: Receive timeout occurred
> [+] Sending WSC NACK
> [!] WPS transaction failed (code: 0x02), re-trying last pin
> [+] Trying pin "12345670"
> [+] Sending authentication request
> [+] Sending association request
> [+] Associated with xx:xx:xx:xx:xx:xx (ESSID: )
> [+] Sending EAPOL START request
> [+] Received identity request
[... 1515 lines cut ...]
> [+] Received identity request
> [+] Sending identity response
> [+] Received deauth request
> [+] Received identity request
> [+] Sending identity response
> [+] Received identity request
> [+] Sending identity response
> [+] Received identity request
> [+] Sending identity response
> [+] Received deauth request
> [!] WARNING: Receive timeout occurred
> [+] Sending WSC NACK
> [!] WPS transaction failed (code: 0x02), re-trying last pin
> [+] Trying pin "12345670"
> [+] Sending authentication request
> [+] Sending association request
> [+] Associated with xx:xx:xx:xx:xx:xx (ESSID: )
> [+] Sending EAPOL START request
> 

BTW, in the router setting, it is currently PBC mode, but there is no option to use the WPS button, it is only the PIN code. Check this pics:
[image]
[image]

Nothing will not change anything if I choose PIN instead

soxrok2212, 2024-05-06:
Try the `-N` flag

reaveruser, 2024-05-06:
@soxrok2212 I tried -N -S -D with all times period, and nothings work. I even reset my router many times, and nothing works.

I am thinking a way to take advantage of the WIFI password as I know it?

soxrok2212, 2024-05-06:
The only option you have there is to log into the web UI. The WPS spec doesn't allow for utilizing the key in any way to get the PIN.

reaveruser, 2024-05-06:
These router manufactures are writing PINs and sometimes the Admin password only in the sticker, if the sticker is gone, no way to find it.
I have many ZTE router with UI passwords are not Admin and I can't access them and no way to access them.

This router is huawei  and the PIN is written on the sticker, but it is gone, no way to find it except of reaver.

It is like these manufactures are doing it on purpose to make us buy new routers if the stickers are gone.

It is sad that there is no way. I hope someone can think of a way to take advantage for the knowing the WIFI password and authenticate using it for getting the PIN code back

Thanks

rofl0r, 2024-05-06:
missing the pin doesn't seem like a big deal to me. you could try oneshot which uses wpa_supplicant, and (if it works at all) work your way forward one pin at a time, by wrapping it into some script. or check if there's any vulnerability for the router that would allow you to get to the raw configuration data using the web interface or other TCP services.

closed 2024-05-06 by soxrok2212