t6x/reaver-wps-fork-t6x #383: master branch can't be compiled on Mac

aelam, 2024-07-26:
I get `timer` related errors 

```log
$ make
cp lwe/wireless.21.h lwe/wireless.h
gcc -DCONF_DIR='"/usr/local/var/lib/reaver"' -Wall -g -O2 -Werror-unknown-warning-option -Wno-unused-but-set-variable -Wno-unused-variable -Wno-unused-function -Wno-pointer-sign -Ilibwps -I. -Ilwe -DCONFIG_IPV6   -c -o globule.o globule.c
warning: unknown -Werror warning specifier: '-Werror-unknown-warning-option' [-Wunknown-warning-option]
warning: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'? [-Wunknown-warning-option]
In file included from globule.c:34:
./globule.h:93:2: error: unknown type name 'timer_t'; did you mean 'time_t'?
        timer_t timer_id;       /* The ID of timer */
        ^~~~~~~
        time_t
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types/_time_t.h:31:33: note: 'time_t' declared here
typedef __darwin_time_t         time_t;
                                ^
2 warnings and 1 error generated.
make: *** [globule.o] Error 1
```

soxrok2212, 2024-07-26:
macOS is not a supported target.

closed 2024-07-26 by soxrok2212

aelam, 2024-07-27:
@soxrok2212 
it stopped working for mac since just the last second commit 4091bf2b31bf9fa7f1cfc913982a525f787ef990 because of just a timer. I think it's not a worthy move

rofl0r, 2024-07-27:
afaik on apple hw there was only one wifi adapter *ever* that supported proper monitor mode and packet injection in 2014-ish intel macbooks, that's why mac support was added back then.