t6x/reaver-wps-fork-t6x #386: [!] WPS transaction failed (code: 0x04), re-trying last pin

skypean, 2024-12-29:
I keep getting the WPS transaction failed, but don't know how to fix it. Please help! 
```
sudo ./reaver -b D4:9A:A0:77:B3:F9 -i wlan0mon -vv -p 37804701 -N

Reaver v1.6.6-git-24-g926eb14 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Waiting for beacon from D4:9A:A0:77:B3:F9
[+] Switching wlan0mon to channel 6
[+] Received beacon from D4:9A:A0:77:B3:F9
[+] Vendor: AtherosC
[+] Trying pin "37804701"
[+] Sending authentication request
[+] Sending association request
[+] Associated with D4:9A:A0:77:B3:F9 (ESSID: Hieu 2.4/5G)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[!] WARNING: Receive timeout occurred
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin "37804701"
[+] Sending authentication request
[+] Sending association request
[+] Associated with D4:9A:A0:77:B3:F9 (ESSID: Hieu 2.4/5G)
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received WSC NACK (reason: 0x000F)
[!] WARNING: Detected AP has WPS setup locked!
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x04), re-trying last pin
[+] Quitting after 0 crack attempts
[-] Failed to recover WPA key
[+] Nothing done, nothing to save.
```
The `wash` command still show that the AP have WPS unlocked
```
└─$ sudo wash -i wlan0mon
BSSID               Ch  dBm  WPS  Lck  Vendor    ESSID
--------------------------------------------------------------------------------
D4:9A:A0:77:B3:F9    6  -42  2.0  No   AtherosC  Hieu 2.4/5G
```

The driver I'm usingi
```
─$ sudo airmon-ng            

PHY	Interface	Driver		Chipset

phy2	wlan0		ath9k_htc	Qualcomm Atheros Communications AR9271 802.11n
```

rofl0r, 2024-12-29:
try oneshot with a single pin to see whether WPS generally works on the router. most router producers simply killed the feature during the last years. killed as in: the transaction stops in the middle, rather than completely disabled.

skypean, 2025-01-04:
I also try oneshot with a single PIN. But it got the same error.

rofl0r, 2025-01-04:
well, if not even wpa_supplicant (used by oneshot) can make a WPS transaction, it means the WPS on the router is broken, so nothing can be done.

closed 2025-01-04 by rofl0r